smit-happens/YCP_EVOS #22: Power-On Reset (POR) and Brownout Detection

PatHock, 2018-02-22:
The Teensy's Kinetis microprocessor has built-in hardware to reset the Teensy during undervoltage conditions. These conditions prevent the Teensy in the Main Brain circuit from powering up reliably - we know that this is an issue. 

Also, the Low Voltage Detect functionality can trigger the Teensy to enter the shutdown stage in the event that the Teensy loses power. This will become more useful later when implementing SD card Logging.

PatHock, 2018-02-23:
This will likely be taken care of in hardware.

Writing to the power mode protect register must be done with "supervisor" privileges. See ARM docs, or kinetis datasheet at around page 104. The power mode protect reg (System Mode Controller, SMC 0x4007_E000) is in a location in flash that is protected etc. The datasheet gives some pointers on how to get around that.

The Power Management Controller (PMC) is in the same Address space (0x4007_D000)

Do more research

PatHock, 2018-03-09:
This issue has been taken care of by the use of a Power On Reset supervisor device (MIC803) on the Main Brain PCB. Solving it in software would take too much time.

closed 2018-03-09 by PatHock

stefanhoelzl, 2019-05-06:
> The Teensy's Kinetis microprocessor has built-in hardware to reset the Teensy during undervoltage conditions. These conditions prevent the Teensy in the Main Brain circuit from powering up reliably - we know that this is an issue.

Hi, 
do you haven any Sources where I can find more informations about the Teensy not powering up reliably caused by undervoltage conditions?

I am also using a Kinetis microcontroller and having sometimes issues when powering up with enabled brownout-detection.

Thank you!

PatHock, 2019-05-19:
> do you haven any Sources where I can find more informations about the Teensy not powering up reliably caused by undervoltage conditions?

Unfortunately I do not. Have you tried checking the power rails on your device with an oscilloscope to see what the voltage looks like when you have power-up issues? Maybe you can make the brownout detection more "aggressive" by changing the configuration. I never had a chance to give the built-in brownout detection a try.

Hope that helps!