smit-happens/YCP_EVOS #25: Gas Pedal polling

smit-happens, 2018-02-22:
Involves working on the PedalController to use the GasPedal Model to grab the analog value and store it in the model

smit-happens, 2018-02-26:
implemented the model for this, need to work on the controller, StageManager, and main to finish this up

PatHock, 2018-03-13:
We can either use a lookup table to correlate the linear and logarithmic pots, or we can calculate the pedal position using an equation that represents the "curves" for each potentiometer. The Teensy 3.6 has a floating point unit which will facilitate floating point calculations, making the calculation option practical.

If calculations are to be used (or even the lookup table), we need to choose our potentiometers soon.

PatHock, 2018-03-13:
EV 3.5.4
All acceleration control signals (between the accelerator pedal and the motor controller) must
have error checking.
For analog acceleration control signals, this error checking must detect open circuit, short to
ground and short to sensor power.

**There is no mention of using both linear and logarithmic pots in the Hybrid Rules**

commit 88fcc132 (2018-03-27): verified working with Unitek and TSCB
fix #129, fix #25
closed 2018-03-27 by smit-happens