smit-happens/YCP_EVOS #55: Orion CAN messages failing

smit-happens, 2018-03-09:
@ShawnPierpont  add to this

ShawnPierpont, 2018-03-12:
Look into having the BMS poll rather than polling the BMS. Then maybe we will only need to read the CAN to get the messages rather than sending a poll to the BMS for the values we need.

smit-happens, 2018-03-16:
@ShawnPierpont please close this with resolution description

ShawnPierpont, 2018-03-17:
After reviewing the pinout of the Teensy, I realized we were using CAN1 rather than CAN0, after modifying my code, I was able to successfully read messages off of the CAN. The BMS is also now configured to periodically send out messages that contains various details of the battery pack such as voltage, current, and state of charge that we can pick up with the Teensy. The next step is now reading only the specific messages that we are interested in, but for this issue, the communication is now set.

PR #137 (2018-03-24): Shawn bms test
Fixes #55

## Proposed Changes

  - ended up being a testing project for FlexCan filtering that Shawn or myself couldn't figure out
  - Got CAN port 2 working on the BMS instead
closed 2018-03-17 by ShawnPierpont