smit-happens/YCP_EVOS #94: Batteries below 20% charge

atokosch, 2018-03-18:
## End goal

## Relevant Info/Links

#### Your Code

```c++
If relevant, paste all of your challenge code in here
```

#### Screenshot/Media

atokosch, 2018-03-18:
Related to #32 , need to verify that the batteries are not below 20%? charge. 

Also related to #92 , if below 20%, then shutdown.

smit-happens, 2018-03-20:
This will most likely be one of the stretch goals for MS2 at the current rate of development

smit-happens, 2018-03-26:
@ShawnPierpont OrionController function for implementation

ShawnPierpont, 2018-04-16:
Need to determine if the handling of this is sufficient and what we want to do when the battery is right above 20% to warn the driver of it approaching.

At the moment I have:
>20% and <=25% = comment about doing a display change on GLCD
>15% and <=20% = set the RPM max limit to 1000 to conserve power, this will serve to also notify the user strongly that the battery is low which is issue #141. 
<=15% = the car enters the shutdown stage and the tractive system is turned off. The GLV is still on.

ShawnPierpont, 2018-04-23:
We still need to test this functionality with the car, but the implementation is now completed. There is a factor stored in the Unitek model called speedCalculationFactor, which is then used in the speed calculation. 
`float UnitekController::calculateRpm(int speedValue)
{
    float percentage = (float)speedValue / unitekModel->MAX_VALUE;
    percentage = percentage / unitekModel->getSpeedCalculationFactor();
    return percentage * unitekModel->getRpmLimit();
}`

It was intended to be a simple shift operation but there were data type issues, so it is a division now. The value is initialized in the UnitekController init function:
`void UnitekController::init(void)
{
    unitekModel = new Unitek();
    //set the speedcalculation factor to not modify the value at all
    unitekModel->setSpeedCalculationFactor(1);
}`

The only time this value is updated after this is in the processOrion function when the batteries reach 20% charge, at 15% the car will go into the shutdown stage and the tractive system will be turned off.

The handling of the state of charge is done in processOrion in StageManager:
`    //first check for pack state of charge issues
    float packSOC = OrionController::getInstance()->getStateOfCharge();
    //maxCellTemperature represents the maximum temperature that we will allow a battery to reach before shutting off the car
    uint8_t maxCellTemperatureCelcius = 60;

    //if the pack is greater than 20% and less than 25%, change the view on the GLCD to notify the driver
    if(packSOC > 20 && packSOC <= 25)
    {
        //the GLCD will update with a change on the user interface for the driver to know the battery is getting low
        //GLCD_warnSocLow()
    }
    //if the pack is greater than 15% and less than 20%, cut the max RPM to 1000 to conserve power and be able to return to the original location without pulling as much power
    else if(packSOC > 15 && packSOC <= 20)
    {
        //set the speed calculation factor to 8 which will make the max rpm sent be 6600/8 = 825 rpms
        UnitekController::getInstance()->storeSpeedCalculationFactor(8);
    }
    //if the pack is at 15%, shut off the tractive system and continually flash the Orion error light until the GLV system is shut off
    else if(packSOC <= 15)
    {
        Logger::getInstance()->log("ORION", "Battery SOC too low, shutting off tractive system", MSG_LOG);
        //shut down the car immediately
        StageManager::shutdown();
    }
    else
    {
        //battery is still at a good charge percentage so there is nothing to do
    }`