smit-happens/YCP_EVOS #106: Brake and Accelerator Pressed Simultaneously

atokosch, 2018-03-19:
## End goal
If need to worry about brake and accel pedal being pressed at same time, maybe use an interrupt for brake so that way it stops receiving input from accel pedal. 

## Relevant Info/Links
#75 gas pedal
Possibly #25 

## Questions
Need to verify that is the brake and pedal are pressed at the same time then it can possibly stall the motor. I know this is an issue with electric race carts, not sure it it will be an issue for us. Will look into with Unitek. Also, this may not be an issue with we declare that you need to only use one foot with the left foot not being used at all.

smit-happens, 2018-03-20:
This won't be accomplished for MS2 but it's something to be decided on for MS3

smit-happens, 2018-05-04:
This is fixed

closed 2018-05-04 by smit-happens