smit-happens/YCP_EVOS #130: Gas Pedal Tolerance

YoungMaker, 2018-03-21:
## End goal
Filter out noise in the gas pedal input so that you're not constantly sending accelerate commands to the Unitek

## Relevant Info/Links
The gas pedal input drifting slightly shouldn't cause any unintended acceleration.

smit-happens, 2018-04-09:
Needs to account for the fractional percent that's being sent hen we are at 0

smit-happens, 2018-04-23:
we still have to configure this based on the position of the potentiometers in the car

PatHock, 2018-06-15:
Added 1uF capacitors to the analog inputs on the main brain PCB, for the accelerator and brake pedals. Noise is mostly eliminated, though there is slight flickering on the brake light, with threshold of 3%. Wiggling the pedal without engaging the brakes is enough to turn the brake light on and off.

smit-happens, 2018-06-15:
fixed in 989b98888e6be2c0761601dfe5ad5beaa7ada736

PR #164 (2018-04-29): #130 #106 found values for the min/max pedal values
Fixes #130 #106 

## Proposed Changes

  - max raw pedal values found so the calculate percent functions work properly for gas/brake
closed 2018-06-15 by smit-happens