smit-happens/YCP_EVOS #135: Multiple task flags going off for buttons

smit-happens, 2018-03-22:
## End goal
Fix the buttons triggering multiple task flags

## Relevant Info/Links
 - Whenever a button is pressed, it will sometimes trigger multiple task flags
 - need to check with a scope first to verify hardware works

smit-happens, 2018-03-22:
Probed with scope, found that pressing one button will cause a spike to occur on another line.

Shown below is a spike above `HIGH` signal voltage level occurring on a button line that wasn't pressed: [image]

smit-happens, 2018-03-23:
@PatHock is fixing this, he'll close it when it's done

PatHock, 2018-03-23:
Fixed in hardware. Button bouncing caused crosstalk, which has been minimized by addressing the bouncing issue.

closed 2018-03-23 by PatHock