smit-happens/YCP_EVOS #141: Notify Driver HV Batteries Below 20% - (driver doesn't look at GLCD)

atokosch, 2018-03-26:
## End goal
Notify driver that the batteries are getting low. Can't use RTD sound.

Ideas:
-Slow car down if GLCD is ignored.
-Blink a light?
-Turn on and off GLCD backlight or different colors

Please Post other ideas below:

atokosch, 2018-03-26:
If get to 15% cut down to 50% power
If get to 10% then cut the car off without the driver doing anything.

atokosch, 2018-03-28:
related to #94

smit-happens, 2018-04-09:
@ShawnPierpont probably based off the cell voltage average, this might be implemented in the Orion controller

ShawnPierpont, 2018-04-16:
Implemented by setting the RPM limit very low, the driver will clearly notice this as a sign the car is low on charge

Implemented in stage manager based off of the state of charge since the state of charge seems to be more accurate at lower percentages, it seems to only act strange right after being charged, but steadies out after some discharging.

closed 2018-04-16 by ShawnPierpont